odwdinc/DWIN_T5UIC1_LCD #1: 'DWIN_LCD' object has no attribute 'checkkey'

Samu1808, 2021-04-10:
I made everything of the setup and connected the Display properly to the Raspberry Pi.

The Display doesnt show anything (but the Backlight is on) and when i spin or push the button this is the output of run.py:

DWIN handshake
Traceback (most recent call last):
  File "/home/pi/DWIN_T5UIC1_LCD/encoder.py", line 47, in transitionOccurred
    self.callback(self.value)
  File "/home/pi/DWIN_T5UIC1_LCD/dwinlcd.py", line 2252, in encoder_has_data
    if self.checkkey == self.MainMenu:
AttributeError: 'DWIN_LCD' object has no attribute 'checkkey'
Traceback (most recent call last):
  File "/home/pi/DWIN_T5UIC1_LCD/encoder.py", line 38, in transitionOccurred
    self.callback(self.value)
  File "/home/pi/DWIN_T5UIC1_LCD/dwinlcd.py", line 2252, in encoder_has_data
    if self.checkkey == self.MainMenu:
AttributeError: 'DWIN_LCD' object has no attribute 'checkkey'
Traceback (most recent call last):
  File "/home/pi/DWIN_T5UIC1_LCD/dwinlcd.py", line 2252, in encoder_has_data
    if self.checkkey == self.MainMenu:
AttributeError: 'DWIN_LCD' object has no attribute 'checkkey'

My setup:
Raspberry Pi 3
Octoprint Version 1.5.3
Klipper  v0.9.1

odwdinc, 2021-04-11:
Looks like it not finding the display
aka `DWIN handshake`, but no `DWIN OK`

May need to change your serial port for Raspberry Pi 3 as this was tested on Raspberry Pi zero W.

looking at https://www.electronicwings.com/raspberry-pi/raspberry-pi-uart-communication-using-python-and-c

"Hardware UART port i.e. GPIO14(TXD) and GPIO15 (RXD) is also known as serial0"

try changing line

`LCD_COM_Port = '/dev/ttyAMA0'`

To:

`LCD_COM_Port = '/dev/serial0'`

Samu1808, 2021-04-12:
Ah this might be the problem, but your suggesstion doesnt fix it:

[image]

As far as I can tell, it also dont make a difference bc serial0 points also to ttyAMA0

I also found out that on rpi3 you have to add 'dtoverlay=pi3-disable-bt' to /boot/config.txt but this also wasn't the problem.
Sadly I'm not very familiar with these basic hardware things as Serial connection and so on.
Do you have any suggestion to find the problem, maybe some tests?

Hajto, 2021-07-21:
Having the same issue. Has anyone found a solution?

Hajto, 2021-07-21:
Flashed original firmware and no luck.

sdavis175, 2021-11-21:
Running into same issue, RPi 4 and went straight from stock to Klipper. Only see DWIN handshake in start, whenever I rotate the knob I get: 

Traceback (most recent call last):
  File "/home/pi/DWIN_T5UIC1_LCD/encoder.py", line 47, in transitionOccurred
    self.callback(self.value)
  File "/home/pi/DWIN_T5UIC1_LCD/dwinlcd.py", line 2252, in encoder_has_data
    if self.checkkey == self.MainMenu:
AttributeError: 'DWIN_LCD' object has no attribute 'checkkey'
Traceback (most recent call last):
  File "/home/pi/DWIN_T5UIC1_LCD/encoder.py", line 38, in transitionOccurred
    self.callback(self.value)
  File "/home/pi/DWIN_T5UIC1_LCD/dwinlcd.py", line 2252, in encoder_has_data
    if self.checkkey == self.MainMenu:
AttributeError: 'DWIN_LCD' object has no attribute 'checkkey'
Traceback (most recent call last):
  File "/home/pi/DWIN_T5UIC1_LCD/encoder.py", line 38, in transitionOccurred
    self.callback(self.value)
  File "/home/pi/DWIN_T5UIC1_LCD/dwinlcd.py", line 2252, in encoder_has_data
    if self.checkkey == self.MainMenu:
AttributeError: 'DWIN_LCD' object has no attribute 'checkkey'

Also using fluidd instead of octoprint, but I don't believe that is the issue since the LCD cannot connect. The LCD is receiving power.

jacrify, 2022-02-01:
I have the same issue. I get DWIN handshake, but no DWIN OK. Looking at the code this means it's getting no response from the lcd board over serial. However moving the encoder wheel registers events and I get the error the OP quoted. 

I've triple checked all my wiring and tested my cable. If any of the original posters found a solution to this, I'd love to hear it.

Samu1808, 2022-03-17:
Hey together, I found a fork `https://github.com/bustedlogic/DWIN_T5UIC1_LCD` which worked for me, I hope for you too.
Enjoy!

sdavis175, 2022-03-18:
I tested with that fork and I am still getting the same checkkey error. LCD has power, what should I be looking to test?

jacrify, 2022-03-18:
Check that the pins on your pi haven’t been shorted. That was the problem
with mine: somehow I blew the pins.

Felos9001, 2024-02-16:
> I tested with that fork and I am still getting the same checkkey error. LCD has power, what should I be looking to test?
> 
> EDIT: This fix did the trick:
> 
> > Looks like it not finding the display aka `DWIN handshake`, but no `DWIN OK`
> > May need to change your serial port for Raspberry Pi 3 as this was tested on Raspberry Pi zero W.
> > looking at https://www.electronicwings.com/raspberry-pi/raspberry-pi-uart-communication-using-python-and-c
> > "Hardware UART port i.e. GPIO14(TXD) and GPIO15 (RXD) is also known as serial0"
> > try changing line
> > `LCD_COM_Port = '/dev/ttyAMA0'`
> > To:
> > `LCD_COM_Port = '/dev/serial0'`

Didnt work sadly. I am having the same issue.

janders13, 2024-07-10:
I had this issue because I followed a tutorial on reddit with wiring diagram that was wrong for me. Look at the markings on the display PCB directly and make sure they go to the right GPIO pins. Or try running it without ENT, A, and B connected.

moffatted, 2024-12-17:
I know this is old, but I have a PI 4 B and this worked for me  LCD_COM_Port = '/dev/serial0' which is linked to ttyS0