odwdinc/DWIN_T5UIC1_LCD #2: Mainboard connection instead of Rpi GPIO

ihrapsa, 2021-04-11:
Hey, great project!
Just wanted to ask how hard would it be to make the display communicate with klipper through the mainboard instead of the Rpi? I know someone accomplished this for the [T5UID1](https://github.com/Desuuuu/DGUS-reloaded-Klipper) touchscreen display that comes with Ender 5 plus, Ender 6, CR10S pro. Are you planning to test this approach on the T5UIC1? I think that would be awesome.

odwdinc, 2021-04-11:
I am not lookin to at this time, I may look to supporting https://github.com/jordanruthe/KlipperScreen

Current setup:

- Run Ender 3 v2 LCD cable to a Raspberry Pi Zero W for UART and power. 
- No usb to printer :)
- Use a USB to UART to run to display.
- This all fits in the case of the display.

ihrapsa, 2021-04-11:
Thanks for replying! I was looking for a way to use the display with klippy running on a linux box that doesn't have GPIOs exposed. It only has 2 USB ports and a serial intarface. What are `Ent`,  `A` and `B` for btw?

ihrapsa, 2021-04-11:
Really like your idea with the Rpi 0W setup nonetheless!

odwdinc, 2021-04-11:
Its for the encoder, aka the knob on the display.