odwdinc/DWIN_T5UIC1_LCD #3: Remove octoprint dependency, and communicate only with moonraker API

JohanAR, 2021-04-11:
Any plans to change the program to use the moonraker API directly? The octoprint dependency is a bit inconvenient for those of us who use different klipper frontends

odwdinc, 2021-04-11:
It’s untested but it should support already the moonraker API directly.
Just use the api key form moonracker vs octoprint.

JohanAR, 2021-04-11:
ok! Just found out about the project so I haven't had a chance to test it yet, only browsed through the code a bit

Xploder, 2021-04-15:
Setting the port to the moonraker service doesn't work and just throws errors.

JohanAR, 2021-04-16:
> Setting the port to the moonraker service doesn't work and just throws errors.

do you have [octoprint_compat] in your moonraker config?

Xploder, 2021-04-16:
> > Setting the port to the moonraker service doesn't work and just throws errors.
> 
> 
> 
> do you have [octoprint_compat] in your moonraker config?

Yes I do

JohanAR, 2021-04-30:
@odwdinc finally got around to testing it, but unfortunately I can confirm what @Xploder was saying, that it doesn't work with Moonraker only

```
DWIN handshake
DWIN OK.
http://127.0.0.1:7125
Waiting for connect to /tmp/klippy_uds

Connection.

Boot looks good
Testing Web-services
Web site exists
Traceback (most recent call last):
  File "/home/johan/DWIN_T5UIC1_LCD/printerInterface.py", line 337, in getREST
    return json.loads(d)
  File "/usr/lib/python3.7/json/__init__.py", line 348, in loads
    return _default_decoder.decode(s)
  File "/usr/lib/python3.7/json/decoder.py", line 337, in decode
    obj, end = self.raw_decode(s, idx=_w(s, 0).end())
  File "/usr/lib/python3.7/json/decoder.py", line 355, in raw_decode
    raise JSONDecodeError("Expecting value", s, err.value) from None
json.decoder.JSONDecodeError: Expecting value: line 1 column 1 (char 0)

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "./run.sh", line 13, in <module>
    API_Key
  File "/home/johan/DWIN_T5UIC1_LCD/dwinlcd.py", line 322, in __init__
    self.pd.init_Webservices()
  File "/home/johan/DWIN_T5UIC1_LCD/printerInterface.py", line 356, in init_Webservices
    ppp = self.getREST('/api/printerprofiles/_default')
  File "/home/johan/DWIN_T5UIC1_LCD/printerInterface.py", line 338, in getREST
    except JSONDecodeError:
NameError: name 'JSONDecodeError' is not defined
^CException ignored in: <module 'threading' from '/usr/lib/python3.7/threading.py'>
Traceback (most recent call last):
  File "/usr/lib/python3.7/threading.py", line 1281, in _shutdown
    t.join()
  File "/usr/lib/python3.7/threading.py", line 1032, in join
    self._wait_for_tstate_lock()
  File "/usr/lib/python3.7/threading.py", line 1048, in _wait_for_tstate_lock
    elif lock.acquire(block, timeout):
KeyboardInterrupt
Shuting down Klippy Socket
```

ihrapsa, 2021-05-22:
Yup! I'm having the same issue. Moonraker only support would be much appreciated.

0madem, 2021-07-01:
Hi, nothing new? How to fix trouble with Moonraker standalone?

ihrapsa, 2021-07-14:
> Hi, nothing new? How to fix trouble with Moonraker standalone?

Not yet unfortunately. Apparently this only works through Octoprint and not Moonraker. Moonraker has an emulation layer for octoprint but it's very basic and mainly developed for slicer upload functionality. We would either have to wait for a direct Moonraker support on this or for the moonraker's octoprint emulation layer to get more functionality.

bharatpareek, 2021-08-22:
So I tried this with Fluiddpi setup.

The code calls the API `/api/printerprofiles/_default` to get the printer information like model, volume, width, height etc. 

But this gived 404 error when using Moonracker directly (even with Octoprint Compat in the config).

Moonracker API endpoint will be this one:  /api/printerprofiles

But even this does not work as we do not get the required properties from the API.

For example:

Moonracker API: 

**List Printer profiles**

HTTP request:

```
GET /api/printerprofiles

Returns:

An object containing simulates Octoprint Printer profile

{
    "profiles": {
        "_default": {
            "id": "_default",
            "name": "Default",
            "color": "default",
            "model": "Default",
            "default": true,
            "current": true,
            "heatedBed": true,
            "heatedChamber": false
        }
    }
}
```

Where as the Octoprint API:

```
GET /api/printerprofiles

{
  "profiles": [
    {
      "id": "_default",
      "name": "Default",
      "color": "default",
      "model": "Generic RepRap Printer",
      "default": true,
      "current": true,
      "resource": "http://example.com/api/printerprofiles/_default",
      "volume": {
        "formFactor": "rectangular",
        "origin": "lowerleft",
        "width": 200,
        "depth": 200,
        "height": 200
      },
      "heatedBed": true,
      "heatedChamber": false,
      "axes": {
        "x": {
          "speed": 6000,
          "inverted": false
        },
        "y": {
          "speed": 6000,
          "inverted": false
        },
        "z": {
          "speed": 200,
          "inverted": false
        },
        "e": {
          "speed": 300,
          "inverted": false
        }
      },
      "extruder": {
        "count": 1,
        "offsets": [
          {"x": 0.0, "y": 0.0}
        ]
      }
    }
}
```

So to workaround this with my Ender 3 V2, I removed the reference of those API properties and added hardcoded values for the time being until I can get these values from API

printerinterface.py

Line no. 358

		```
self.MACHINE_SIZE = "{}x{}x{}".format(
			int(235),
			int(235),
			int(250)
		)
		self.X_MAX_POS = int(230)
		self.Y_MAX_POS = int(230)
```

bharatpareek, 2021-08-22:
My display was stuck on Creality logo but now it turns on, shows the menu entries too, but no option work at all. Seems like I will have to modify all the non working API endpoints to make it work with Fluiddpi & Moonracker.

ihrapsa, 2021-08-22:
@bharatpareek Check out this fork: https://github.com/bustedlogic/DWIN_T5UIC1_LCD
He seems to have fixed it.

bharatpareek, 2021-08-22:
> @bharatpareek Check out this fork: https://github.com/bustedlogic/DWIN_T5UIC1_LCD
> He seems to have fixed it.

Thank you so much, I thought I ll have to do everything from scratch. 😁

Thank you @bustedlogic

bustedlogic, 2021-08-22:
> > @bharatpareek Check out this fork: https://github.com/bustedlogic/DWIN_T5UIC1_LCD
> > He seems to have fixed it.
> 
> Thank you so much, I thought I ll have to do everything from scratch. 😁
> 
> Thank you @bustedlogic

You're welcome and I hope it works well for you. I ported it to moonraker so it works with every frontend. tried to fix all the bugs as well as, but I'm sure I missed one or two.

Quick note, my version is setup for a reverse direction controller knob (an ender3v2 clone). The only difference is in the run.py (created yourself from the readme). Change the `19, 26 `to `26, 19` on line 4, or just copy run.py from the original readme

bharatpareek, 2021-08-23:
Yes will do, I was thinking if we can remove the LCD from the LCD mount of Ender 3 v2, and use it directly with DSI port on board. :D  

But somewhere I read that RPI has some closed source code which is needed for the DSI port. 

But anyway, the main issue is solved. :)

ihrapsa, 2021-08-23:
> Yes will do, I was thinking if we can remove the LCD from the LCD mount of Ender 3 v2, and use it directly with DSI port on board. :D
> 
> But somewhere I read that RPI has some closed source code which is needed for the DSI port.
> 
> But anyway, the main issue is solved. :)

That would probably require removing the lcd panel from the T5 driver pcb board and using a custom one that could interface with HDMI or DSI. It's not that simple especially if you don't have access to the LCD datasheet. I've seen a hobby [project](https://hackaday.com/2014/11/02/using-cell-phone-screens-with-any-hdmi-interface/) of driving smartphone LCD panels With HDMI through a custom board which obviously shows that this is not impossible. However, afaik manufacturers don't really have a standard LCD connector and that just complicates things even more. 

Even if you manage to drive the lcd through HDMI/DSI it's still kind of useless without a touchscreen digitizer so you'd still have to use an encoder or some sort of HID to control it.

ntcong, 2021-09-05:
I added some compatibility with moonraker here. So far things are working
https://github.com/odwdinc/DWIN_T5UIC1_LCD/pull/8

bradvuv, 2021-11-06:
Hi guys, I'm using Fluiddpi. I can't find the file klipper.service in ect/default to enable the API socket. 

"Enabling Klipper's API socket
By default, the Klipper's API socket is not enabled. In order to use the API server, the file /etc/default/klipper need to be updated form

KLIPPY_ARGS="/home/pi/klipper/klippy/klippy.py /home/pi/printer.cfg -l /tmp/klippy.log"
To:

KLIPPY_ARGS="/home/pi/klipper/klippy/klippy.py /home/pi/printer.cfg -a /tmp/klippy_uds -l /tmp/klippy.log"

Everything went smoothly and the LCD boots only it won't connect to mailsail/fluidd due to the API socket issue.

Any help would be great.Thanks in advance