odwdinc/DWIN_T5UIC1_LCD #11: DWIN_T5UIC1_LCD/printerInterface.py - KeyError: 'model'

f0o, 2022-02-09:
Not sure what to do about this one...

```
# python3 run.py

DWIN handshake
DWIN OK.
http://127.0.0.1:80
Waiting for connect to /tmp/klippy_uds

Connection.

Boot looks good
Testing Web-services
Web site exists
Traceback (most recent call last):
  File "run.py", line 13, in <module>
    API_Key
  File "/home/pi/DWIN_T5UIC1_LCD/dwinlcd.py", line 322, in __init__
    self.pd.init_Webservices()
  File "/home/pi/DWIN_T5UIC1_LCD/printerInterface.py", line 357, in init_Webservices
    self.SHORT_BUILD_VERSION = ppp['model']
KeyError: 'model'
```

LCD Is now showing that initial Creality Loading Bar indefinitely without animation.

Ideas?

samirthegreat, 2022-02-13:
Okay i found a solution! 
https://github.com/bustedlogic/DWIN_T5UIC1_LCD 
use this fork this uses moonraker api

bobdivx, 2023-04-19:
hello

Hello, I'm stuck
my screen is flashing.
but when I turn the wheel of the screen some lines appear below run.py
what to do please

`pi@raspberrypi:~/DWIN_T5UIC1_LCD $ python3 ./run.py                             
DWIN handshake

^CTraceback (most recent call last):
  File "/home/pi/DWIN_T5UIC1_LCD/./run.py", line 9, in <module>
    DWINLCD = DWIN_LCD(
  File "/home/pi/DWIN_T5UIC1_LCD/dwinlcd.py", line 317, in __init__
    self.lcd = T5UIC1_LCD(USARTx)
  File "/home/pi/DWIN_T5UIC1_LCD/DWIN_Screen.py", line 63, in __init__
    while not self.Handshake():
  File "/home/pi/DWIN_T5UIC1_LCD/DWIN_Screen.py", line 111, in Handshake
    time.sleep(0.1)
KeyboardInterrupt
`

marine1988, 2023-07-23:
same problem here

AttributeError: 'DWIN_LCD' object has no attribute 'checkkey'