odwdinc/DWIN_T5UIC1_LCD #19: Help with pin connection

joelstitch, 2023-06-20:
I want to run this on my Ender 3 V2 but am confused about how to connect the wires to the display. Can someone share a image of how its done? Do they connect to where the original cable hook or connects straight to the board of the screen?

joelstitch, 2023-06-23:
I followed the instructions very carefully but when I get to running the run.py I just get all these errors:

`DWIN handshake
Traceback (most recent call last):
  File "/home/pi/DWIN_T5UIC1_LCD/dwinlcd.py", line 2252, in encoder_has_data
    if self.checkkey == self.MainMenu:
AttributeError: 'DWIN_LCD' object has no attribute 'checkkey'
Traceback (most recent call last):
  File "/home/pi/DWIN_T5UIC1_LCD/dwinlcd.py", line 2252, in encoder_has_data
    if self.checkkey == self.MainMenu:
AttributeError: 'DWIN_LCD' object has no attribute 'checkkey'
Traceback (most recent call last):
  File "/home/pi/DWIN_T5UIC1_LCD/dwinlcd.py", line 2252, in encoder_has_data
    if self.checkkey == self.MainMenu:
AttributeError: 'DWIN_LCD' object has no attribute 'checkkey'
Traceback (most recent call last):
  File "/home/pi/DWIN_T5UIC1_LCD/dwinlcd.py", line 2252, in encoder_has_data
    if self.checkkey == self.MainMenu:
AttributeError: 'DWIN_LCD' object has no attribute 'checkkey'
Traceback (most recent call last):
  File "/home/pi/DWIN_T5UIC1_LCD/dwinlcd.py", line 2252, in encoder_has_data
    if self.checkkey == self.MainMenu:
AttributeError: 'DWIN_LCD' object has no attribute 'checkkey'
Traceback (most recent call last):
  File "/home/pi/DWIN_T5UIC1_LCD/dwinlcd.py", line 2252, in encoder_has_data
    if self.checkkey == self.MainMenu:
AttributeError: 'DWIN_LCD' object has no attribute 'checkkey'
Traceback (most recent call last):
  File "/home/pi/DWIN_T5UIC1_LCD/dwinlcd.py", line 2252, in encoder_has_data
    if self.checkkey == self.MainMenu:
`

Running it as sudo gives me the same errors. Any suggestions on what to do to fix this?

Felos9001, 2024-02-16:
I am having the same issue,
I am running klipper on mainsailos and Moonraker. installed with kiuah.

```
python3 ./run.py

DWIN handshake
Traceback (most recent call last):
  File "/home/felos/DWIN_T5UIC1_LCD/encoder.py", line 38, in transitionOccurred
    self.callback(self.value)
  File "/home/felos/DWIN_T5UIC1_LCD/dwinlcd.py", line 2252, in encoder_has_data
    if self.checkkey == self.MainMenu:
AttributeError: 'DWIN_LCD' object has no attribute 'checkkey'
```